duckietown/Software #151: catkin_make apriltag_ros no BoolStamped.h

YuehChuan, 2016-07-12:
cannot catkin_make under catkin_ws

erro show:

> fatal erro: duckietown_msgs/BoolStamped.h No such file or directory
> #include <duckietown_msgs/BoolStamped.h>
> 
> compilation terminated.

solution : uncomment

`#  duckietown_msgs`

under
  ~/duckietown/catkin_ws/src/apriltags_ros/CMakeList.txt

cmake_minimum_required(VERSION 2.8.3)
project(apriltags_ros)
...
find_package(catkin REQUIRED COMPONENTS
  apriltags
  image_transport
  roscpp
  sensor_msgs
  geometry_msgs
`#  duckietown_msgs`
  tf
  cv_bridge
  message_generation
  nodelet
  std_msgs
  cmake_modules
)

some naughty boy.

EwingKang, 2016-07-30:
Nice job! Thanks for the info! The first thing came up by google. lol
One small sidenote: I believe it should be one layer deeper. 
~/duckietown/catkin_ws/src/apriltags_ros/apriltags_ros/CMakeList.txt

liampaull, 2016-07-30:
hi - would you mind documenting this on ask.duckietown.org ?

YuehChuan, 2016-07-31:
thanks @EwingKang  modifie the path.

YuehChuan, 2016-07-31:
@liampaull
Already add, good idea.
http://ask.duckietown.org/question/18/solvedfatal-erro-duckietown_msgsboolstampedh-no-such-file-or-directory/

liampaull, 2016-08-18:
hi this should be fixed in master.

closed 2016-08-18 by liampaull