duckietown/Software #152: Cannot find the "machines" file

EwingKang, 2016-08-02:
We have followed the [Setup Step 2.00 - From SD image to RC control](https://docs.google.com/document/d/1HP5ao3LwgQ1EkdRb3ksiMg8zdrpJXjSIH_XG2RFHyes/edit?pref=2&pli=1#) to the step of "Add your vehicle to the machines file". Except there is no such file under "~/duckietown/catkin_ws/src/duckietown/machines"

solution:
After successful catkin_make, reboot your duckiebot

> duckiebot $ sudo reboot

Reconnect using SSH, start boybu, source the environment bash

> duckiebot $ source environment.sh

And the file is automatically generated at "~/duckietown/catkin_ws/src/duckietown/machines" 

Note:

1. We're using SSH+byobu all the time.
2.The file generating code is inside the ~/duckietown/Make file and can be called by environment.sh. I don't know how, but it doesn't seem to work if you havn't reboot the duckitbot right after the successful catkin_make.
3.This specific tutorial file is locked, I cannot add suggestion like [Step 1.1](https://docs.google.com/document/d/1qMso-yhvK_y5lQ0qNpvDtPJ7M1iF5p7-l0_lJBTxHqo/edit). It's probably not a good idea to let anyone to edit anyway

EbadSyed, 2016-11-06:
Somehow I cant commit after editing the machine file, any suggestions ?
 git status shows everything is upto date

YuehChuan, 2016-12-10:
There is a Makefile under ~/duckietown
just type make  
$`make`
https://github.com/duckietown/Software/blob/master/Makefile#L10
It  would trigger 
`python setup/create-machines-file.py $(scuderia) > $(machines)`
which will parse what you type  ( duckiebot info) in ~/duckietown/scuderia.yaml
machine file is for remote roslaunch
ref: http://answers.ros.org/question/41413/how-to-include-a-launch-file-on-a-remote-machine-from-a-launch-file/
By the way, in environment.sh
`exec "$@" ` 
is also important to make remote roslaunch work
https://github.com/duckietown/Software/blob/master/environment.sh#L24

AndreaCensi, 2017-08-16:
Closing because the mechanism for machines files has changed in the new version.

closed 2017-08-16 by AndreaCensi