duckietown/Software #162: is the imu being used

cambazz, 2017-05-27:
Hello,

I constructed myself a duckie bot, and got it to do lane following, and it works. well almost :) 

Currently I do not have the IMU board, and found out 

https://github.com/duckietown/Software/blob/master/catkin_ws/src/adafruit_imu/script/adafruit_imu.py

just publishes the data:

`        self.pub_imu = rospy.Publisher("~adafruit_imu", Imu, queue_size=10)
        self.pub_mag = rospy.Publisher("~adafruit_mag", MagneticField, queue_size=10)`

I scanned the source tree further and found that only

~/duckietown/catkin_ws/src/duckietown/launch/old/imu_remote.launch

launches the imu reader. 

So my question is that do we really need the imu board? Does the algorithms use it, or is it for just making observations about the linear velocity and mag?

Thank you for the duckiebot project.

Best Regards,
C.

EbadSyed, 2017-05-31:
Greetings @cambazz ,

Although I am not able to answer your question. But I have also constructed a duckiebot myself and I am on initial stages of trying to improve the lane following and facing many problems.

If possible would you be able to help me out and discuss different things about duckiebot.

Thanks

cambazz, 2017-06-02:
Hello @EbadSyed,

There is http://answers.ros.org where you can ask questions, and I just posted this question http://answers.ros.org/question/263071/duckietown/ asking whether if we can have a duckietown section in answers.ros.org

Best Regards,
C.

closed 2017-08-29 by AndreaCensi