duckietown/Software #163: Incomplete merge

AuthenticEshkinKot, 2017-07-04:
Software/catkin_ws/src/dagu_car/src/wheels_cmd_switch_node.py:8
It looks like merge conflict was not solved.

AndreaCensi, 2017-08-16:
@liampaull  please look at this (in master)

We definitely need unit tests...

liampaull, 2017-08-17:
yes there is a merge not solved. However `wheels_cmd_switch_node` is deprecated and replaced by `car_cmd_switch_node`

AndreaCensi, 2017-08-17:
Let's move the deleted stuff out of the way?

A triage system like the following?

```
catkin_ws/src/polished/
catkin_ws/src/topolish/
catkin_ws/src/deprecated/
```

liampaull, 2017-08-17:
ok but this is a big job and should be done carefully.

AndreaCensi, 2017-08-18:
I started doing this. 

Note that with the exception of the `duckietown` package, you can move the packages around however you want, and nothing will break.

liampaull, 2017-08-21:
yes that's the beauty of the `find` command in ROS. They can even be in nested directories but looks like you figured that out already.

AndreaCensi, 2017-08-21:
It would be very helpful if you went in and removed all obsolete material, or if there is historical interest, preserve it but clearly mark it as non-executable (e.g. old_code.py -> old_code.py.old).
We will have many people working on it; I want to give them a clean codebase.

closed 2017-08-29 by AndreaCensi