duckietown/Software #165: line detector only publishing one line per color max

liampaull, 2017-08-14:
it used to be that the line detector published a huge array of segments but not it only publishes one per color. I think this must be a bug in the line detector node

AndreaCensi, 2017-08-15:
Is this a regression from switching to RPI3 and slightly different OpenCV?

YuehChuan, 2017-08-15:
Mmmm,  as @AndreaCensi  mention the only change in line detector node is   
cv2.CV_LOAD_IMAGE_COLOR to cv2.IMREAD_COLOR
https://github.com/duckietown/Software/commit/3554ea292c660a1e0300d412984b8c1fe7ac2f98

liampaull, 2017-08-15:
I actually have a feeling that this bug is much older. I think it was caused by the integration of the illumination invariance at the end of the Fall2017 semester. 
@YuehChuan if you run the lane following pipeline you can see in the output to the command line how many detections there are for each color. Can you tell me if you see those numbers ever larger than one?

AndreaCensi, 2017-08-15:
Shivers... we really need unit tests.

liampaull, 2017-08-15:
Fixed
https://github.com/duckietown/Software/pull/166/commits/511e44cb1db83c278ad42add3b49b832dd607b97

closed 2017-08-15 by liampaull