duckietown/Software #167: build fails first time on Pi (but ok on laptop)

AndreaCensi, 2017-08-16:
The first time catkin_make is run, the apriltags_ros package fails compilation with the following messages. The second time it compiles time.

Note:

- this only happens on the Pi
- the dependency on `duckietown_msgs` seem to be written correctly 

```
In file included from /home/andrea/duckietown/catkin_ws/src/apriltags_ros/apriltags_ros/src/apriltag_detector.cpp:1:0:
/home/andrea/duckietown/catkin_ws/src/apriltags_ros/apriltags_ros/include/apriltags_ros/apriltag_detector.h:6:41: fatal error: duckietown_msgs/BoolStamped.h: No such file or directory
compilation terminated.
apriltags_ros/apriltags_ros/CMakeFiles/apriltag_detector.dir/build.make:62: recipe for target 'apriltags_ros/apriltags_ros/CMakeFiles/apriltag_detector.dir/src/apriltag_detector.cpp.o' failed
make[3]: *** [apriltags_ros/apriltags_ros/CMakeFiles/apriltag_detector.dir/src/apriltag_detector.cpp.o] Error 1
  1 cmake_minimum_required(VERSION 2.8.3)
make[3]: Leaving directory '/home/andrea/duckietown/catkin_ws/build'
```

YuehChuan, 2017-08-17:
I ever got this issue, but forgot how to solve it. It's related to something missed in apriltags_ros CMakeLists.txt, 
let me check check. By the way this stand alone repo https://github.com/YuehChuan/apriltags_ros 
already successfully built on pi3.

liampaull, 2017-08-17:
My suspicion is that this is  caused by not declaring the message dependency. It fails the first time because the duckietown_msgs are not built yet, but on the next go around they have been built (luckily) during the build process.

This should be solved by adding:
```
generate_messages(
  DEPENDENCIES
  duckietown_msgs
)
```
to the CMakeLists.txt in apriltags_ros. I pushed the change (can't do any harm). 
I don't have my robot right now to test.

YuehChuan, 2017-08-17:
agree, I remember it's because of msg generation issue

AndreaCensi, 2017-08-18:
Fixed.

closed 2017-08-18 by AndreaCensi