duckietown/Software #186: from what-the-duck

liampaull, 2017-09-05:
* Found 1 invalid test(s):

  - INVALID TEST:   This robot is mentioned in scuderia.

Note that these are *bugs* in `what-the-duck` or its plugins and should be reported 
as soon as possible at the tracker: 

 https://github.com/duckietown/Software/issues

AndreaCensi, 2017-09-07:
Hi Liam,

I would need to see the details of the errors. The extended error message should be given before the summary.

liampaull, 2017-09-08:
I think it was caused by the fact that I hadn't built my machines file yet or something like that. I made it go away easily. I was just following the instructions of the message and filing as a bag. Would be hard for me to reproduce now..

AndreaCensi, 2017-09-10:
ok

closed 2017-09-10 by AndreaCensi