duckietown/Software #187: cam_info_reader_node.py crashing on startup

liampaull, 2017-09-05:
File "/home/ubuntu/duckietown/catkin_ws/src/05-teleop/pi_camera/src/cam_info_reader_node.py", line 23
    self.camera_info_msg = None
       ^
SyntaxError: invalid syntax
process[shamrock/image_proc_node-6]: started with pid [13452]
================================================================================REQUIRED process [shamrock/cam_info_reader_node-3] has died!
process has died [pid 13449, exit code 1, cmd /home/ubuntu/duckietown/catkin_ws/src/05-teleop/pi_camera/src/cam_info_reader_node.py cam_info_reader_node/camera_info:=camera_node/camera_info cam_info_reader_node/compressed_image:=camera_node/image/compressed __name:=cam_info_reader_node __log:=/home/ubuntu/.ros/log/0a26140c-9293-11e7-bb35-b827eb13be69/shamrock-cam_info_reader_node-3.log].
log file: /home/ubuntu/.ros/log/0a26140c-9293-11e7-bb35-b827eb13be69/shamrock-cam_info_reader_node-3*.log
Initiating shutdown!
================================================================================

liampaull, 2017-09-05:
it is making me nervous that the basic pipelines are not working 1 day before class starts

liampaull, 2017-09-05:
fixed by https://github.com/duckietown/Software/commit/1097ebcebeb64613775cbd713156bc124608ce17
not merged yet to master cherry-pick if necessary

closed 2017-09-05 by liampaull

YuehChuan, 2017-09-06:
ha good luck, oh i found that I made a wrong pull request