duckietown/Software #210: SLAM

EbadSyed, 2017-10-02:
I want to implement some kind of Visual SLAM algorithm using Duckiebot in Duckietown. Any heads up on where can I start or how any instructions ? I will be very grateful.

@liampaull @AndreaCensi

liampaull, 2017-10-02:
@EbadSyed try Orb SLAM?

EbadSyed, 2017-10-05:
I want to map the lanes of Duckietown. I have not used tiles for my Duckietown instead it is printed on a panaflex sheet. ORB SLAM not working on Kinetic and ORBSLAM2 has no features to build a cost map.

Is there any way to do lane mapping ?

@liampaull @AndreaCensi

EbadSyed, 2017-10-08:
Did managed to build ORBSLAM 2 on Pi but not the ROS Build.
However, I tried ORBSLAM 2 using Duckiebot as Slave and my laptop as master. Camera launch running on Duckiebot and ORBSLAM computation on Laptop. But my network is not allowing me a very good FPS maybe by using better Wlan adapter I can do it much better.

AndreaCensi, 2017-11-26:
Closing because this is not a software defect.

closed 2017-11-26 by AndreaCensi