duckietown/Software #211: Where can we put contributions in hardware development

altineller, 2017-10-03:
Hello,

I am making a number of parts and circuits for the duckiebot. After making number of 3d printed parts for my duckiebot, I designed circuit shaped as a duckiebot bumper, having the PCA9685 and leds soldered in. But before I had the PCB printed, I found out an easier way by means of using an ATTINY85 as an i2c slave, and 5 WS2812 addressable leds. Here is a picture of the circuit in breadboard:

[image]

I used the driver by usedbytes on https://github.com/usedbytes/neopixel_i2c with minor modification.

Then I made a copy of Adafruit_PWM_Servo_Driver as NeoPixel_I2C_Driver, renamed the py file as well, and made some modifications in init, reset, and setPWM methods. Both reset and setAllPWM methods work and have been tested as the original.

`#!/usr/bin/python

import time
import math
from Adafruit_I2C import Adafruit_I2C

# ============================================================================
# NeoPixel_I2C_Driver
# ============================================================================

class PWM :
  # Registers/etc.
  __OFFSET             = 0x04
  __RESET	       = 0x01
  __GLOBAL	       = 0x02

  general_call_i2c = Adafruit_I2C(0x00)

  @classmethod
  def softwareReset(cls):
    "Resets the LEDS by writing 0x01 on the control register"
    cls.general_call_i2c.writeRaw8(self.__RESET)

  def __init__(self, address=0x40, debug=False):
    self.i2c = Adafruit_I2C(address)
    self.i2c.debug = debug
    self.address = address
    self.debug = debug
    self.setAllPWM(0, 0)
    time.sleep(0.01)

  def setPWM(self, channel, on, off):
    "Sets a single PWM channel"
    pwmval = int((on / float(off)) * 255)
    self.i2c.write8(self.__OFFSET + channel, pwmval)
    #CA check this statement
    time.sleep(0.00001)

  def setAllPWM(self, on, off):
    "Sets all PWM channels"
    if off==0 and on==0 :
       pwmval = 0
    elif off==0 :
       pwmval = 255
    else :
    	pwmval = int((on / float(off)) * 255)
    self.i2c.write8(1, pwmval)
    self.i2c.write8(2, pwmval)
    self.i2c.write8(3, pwmval)
    self.i2c.write8(0, self.__GLOBAL)
    self.i2c.write8(0, 0x00)`

I also had to edit the ~/duckietown/catkin_ws/src/40-coordination/rgb_led/include/rgb_led/rgb_led.py 

`    OFFSET_RED   = 1
    OFFSET_GREEN = 0
    OFFSET_BLUE  = 2
`
Basically this is because the i2c code on the attiny expects GRB instead of RGB. 

I also had to edit ~/duckietown/catkin_ws/src/05-teleop/adafruit_driver/setup.py to add NeoPixel_I2C_Driver:

`setup_args = generate_distutils_setup(
    packages=['Adafruit_ADS1x15', 'Adafruit_GPIO','Adafruit_I2C','Adafruit_LSM303','Adafruit_MotorHAT','Adafruit_PWM_Servo_Driver','Gyro_L3GD20', 'NeoPixel_I2C_Driver'],
    package_dir={'': 'include'},
)
`

I have run all the tests, and it seems to be working with the correct order.

I originally wanted to modify the neopixel_i2c to use similar addressing scheme like PCA9685 - so all this can be transparent without requiring any modification in the duckiebot code.

altineller, 2017-10-03:
The minor modification on the https://github.com/usedbytes/neopixel_i2c was to comment the `swirly();` at line 124.

Also, to mount the WS2812 leds on the bumper, I will probably need to change the ordering of the leds from {fl,bl,top,br,fr} to something like {fl,top,fr,br,bl} --and this way minimal cabling will be used.

AndreaCensi, 2017-10-07:
@altineller  You can put these developments in the folder `hardware` of this repository via a pull request.

FYI @liampaull @tanij

altineller, 2017-10-09:
Hello @AndreaCensi 

I have written a more detailed document and draw a circuit outlining the modification done and put it under hardware/ws2812driver. 

I also put the modified driver under ws2812driver/NeoPixel_I2C_Driver folder. I avoided writing files under catkin_ws/src and just carefully described the modification required. I think maybe a build switch can select between the pwm hat and the ws2812 driver in the future.

Here is a video of it working: https://vimeo.com/235226558

Also please see: https://www.thingiverse.com/thing:2535659 for a 3d printed part we designed for the camera.

breandan, 2017-10-10:
> Here is a video of it working: https://vimeo.com/235226558

@altineller I couldn't find anything related to Duckietown in the video URL. Vimeo autoplays one of their featured videos about 10 seconds after a video is finished, updating the URL in the browser. Did you mean to link [one of these](https://vimeo.com/altineller) instead?

altineller, 2017-10-10:
Hello @breandan 

I am sorry, there was a paste error, this should have been the correct url : https://vimeo.com/237480521

closed 2017-10-15 by AndreaCensi