duckietown/Software #223: os.environ doesn't seem to work when launched "remotely"

liampaull, 2017-10-10:
When running a python ros node using the remote protocol (i.e. through ssh), the environment variables do not seem to be loaded correctly by os.environ. 

This is important for finding the local "DUCKIEFLEET_ROOT" environment variable to load and save calibrations. 

For now - the `camera.launch` file should be launched from the robot until this is resolved. 

@duckietown/fall-2017-instructors-and-tas

AndreaCensi, 2017-10-10:
@liampaull rather than "not being loaded correctly by os.environment", this is "the variable DUCKIEFLEET_ROOT is not in the environment". Where is the environment variable set?

AndreaCensi, 2017-10-10:
@liampaull 

I am not sure that ROS is running the equivalent that happens when you run "ssh host". It uses a wrapper. 

I think this is similar to this:
https://stackoverflow.com/questions/19938993/python-using-env-variables-of-remote-host-with-ssh

AndreaCensi, 2017-10-10:
@liampaull 

Environment variables are overrated anyway. 

We can read that piece of information in a different way. I could parse the .bashrc myself.

liampaull, 2017-10-10:
fyi tested with duckietown_utils code and problem persists.

AndreaCensi, 2017-10-11:
@liampaull that was not suppose to fix this problem; it was just to make everything more maintainable.

liampaull, 2017-10-11:
that's good because it didn't :)

AndreaCensi, 2017-10-14:
Workaround that I developed is that if the variable is not set, and there exists a directory `~/duckiefleet`, that directory is used as the duckiefleet root.

closed 2019-05-09 by liampaull