duckietown/Software #256: Can not connected with the motor shield

zqzhang123, 2017-11-06:
After install the ros and create the machine file with hostname ="biecuole" and username = "zhang", there is a problem about the roslaunch. It seems that I can't establish the connection with the hostname.I can ping with the hostname"zhang". Do you haveany ideas?
[image]

YuehChuan, 2017-11-06:
It try to do remote launch 
your pi3 user name is robot2,  the situation would be more  clearly for us if you describe workspace  configuration.
ex:
pi3 user: robot2    vim /etc/hosts    192.168.0.x zhang
Labtop: user: zhang   vim /etc/hosts    192.168.0.x zhang
also provide screenshot of machine file helps.

liampaull, 2017-11-07:
you haven't followed the ssh config instructions properly. This is not an issue.

closed 2017-11-07 by liampaull