duckietown/Software #335: docker/env.sh assumes that wlan0 exists and connected, and fails silently if not

AndreaCensi, 2018-09-06:
In docker/env.sh we have:

```
IP=$(ifconfig wlan0 | grep "inet " | awk -F'[: ]+' '{ print $4 }')
```

if `wlan0` has no address, then IP is set to "" and it is very difficult to debug the problems down the line.

AndreaCensi, 2018-09-06:
@afdaniele another problem that can arise

afdaniele, 2018-09-06:
I hit this before. When you run the container without `network: "host"`, the container does not see `wlan0`. You get a weird error in the initialization of the bash when this happens. I can't remember why I didn't create an issue though. I'm not sure what is the best way to go with this.

afdaniele, 2018-09-06:
How about this?
```
DEV=$(ip r | grep "default via" | head -1 | awk -F'[: ]+' '{ print $5 }')
IP=$(ifconfig $DEV | grep "inet " | awk -F'[: ]+' '{ print $4 }')
```
This will get the device that is associated with the default gateway (`wlan0` for Duckiebot and `eth0` for Watchtowers) and then use that device to get the current IP.

AndreaCensi, 2018-09-06:
looks good to me

AndreaCensi, 2018-09-06:
by the way my problem was different: I don't want the watchtowers to have wlan0 active, so I put invalid wifi credentials and wlan0 has no address

afdaniele, 2018-09-06:
we can decide at flashing time whether a RaspberryPi will have `wlan0` or `eth0` (no need to use invalid credentials), but the code will work nicely on both.

AndreaCensi, 2018-09-06:
@afdaniele I have a dream in which users don't need to flash the sd card, and they can use the same for all raspberry PIs in duckietown.

breandan, 2018-09-18:
You can also use `hostname -I` to obtain the IP address. Meant to submit a PR, but this is how I fixed this.

breandan, 2018-09-18:
Will update PR #346 to use `hostname -I` instead of parsing the output of `ifconfig`.

closed 2018-10-10 by breandan