duckietown/Software #365: ImportError r.e QuickApp Base

bhairavmehta95, 2018-10-17:
When trying to run the software stack out of box (particularly the single image pipeline with the command below):

`rosrun complete_image_pipeline single_image_pipeline`

I get this error stemming from quick app:
```
  File "/home/bhairavm/coding/Software/catkin_ws/src/10-lane-control/complete_image_pipeline/src/single_image_pipeline", line 4, in <module>
    from duckietown_utils.cli import d8app_run
  File "/home/bhairavm/coding/Software/catkin_ws/src/00-infrastructure/duckietown/include/duckietown_utils/cli.py", line 3, in <module>
    from quickapp import QuickApp, QuickAppBase
  File "/home/bhairavm/.local/lib/python2.7/site-packages/quickapp/__init__.py", line 16, in <module>
    from .quick_app_base import *
  File "/home/bhairavm/.local/lib/python2.7/site-packages/quickapp/quick_app_base.py", line 8, in <module>
    from conf_tools.utils import indent
ImportError: No module named conf_tools.utils
```

bhairavmehta95, 2018-10-17:
Fixed by installing `conftools` and `networkx` (which came from a later error)

I also had to install `PyGeometry` even though it was listed in the `requirements.txt`?

Renaming Issue to Dependencies

AndreaCensi, 2018-10-18:
@bhairavmehta95 the dependencies are declared in `requirements.txt`. This is a problem with some Dockerfile somewhere who didn't install them.

What image in particular were you using?

bhairavmehta95, 2018-10-19:
I was trying to install the software repo on my desktop, so no Image.

bhairavmehta95, 2018-10-19:
Yeah I ran the script `dependencies_laptop` which calls `dependencies_common` which calls that.