duckietown/Software #393: duckiebot does not move during wheel calibration, not cause not clear

davesnowdon, 2019-02-04:
I have successfully completed camera calibibration but I’ve been unable to complete the wheels calibration step for my duckiebots. The make hw-test-kinematics test fails and the robot does not move at all. The only error I see is the failure to open the joystick device but I do not know why it would require that. I’ve tried with the duckietown/rpi-duckiebot-joystick-demo:master18 running and not running but the error remains the same.

Here’s the complete output that I see in the shell window. Can anyone shed any light on what the issue might be?

`root@duckiebot2:/home/software# make hw-test-kinematics
Makefiles/Makefile.build.mk:118: warning: overriding recipe for target 'build-pyc-clean'
Makefiles/Makefile.build.mk:43: warning: ignoring old recipe for target 'build-pyc-clean'
# Put here procedures to check the environment is valid
#-./what-the-duck
Testing Kinematics Calibration
bash -c "rostest indefinite_navigation calibrate_kinematics.test veh:=duckiebot2"
... logging to /root/.ros/log/rostest-duckiebot2-776.log
[ROSUNIT] Outputting test results to /root/.ros/test_results/indefinite_navigation/rostest-tests_calibrate_kinematics.xml
[INFO] [1548533313.212423]: [/duckiebot2/camera_node] Initializing......
[INFO] [1548533313.243814]: [/duckiebot2/camera_node] ~framerate_high = 30 
[INFO] [1548533313.273882]: [/duckiebot2/camera_node] ~framerate_low = 15 
[INFO] [1548533313.340555]: [/duckiebot2/camera_node] ~res_w = 640 
[INFO] [1548533313.380899]: [/duckiebot2/camera_node] ~res_h = 480 
[INFO] [1548533313.924926]: [/duckiebot2/camera_node] Initialized.
[INFO] [1548533313.942541]: [/duckiebot2/camera_node] Start capturing.
[INFO] [1548533314.023383]: [/duckiebot2/cam_info_reader_node] ~config = baseline 
[INFO] [1548533314.043567]: [/duckiebot2/camera_node] Published the first image.
[INFO] [1548533314.079280]: [/duckiebot2/cam_info_reader_node] ~cali_file_name = duckiebot2 
[INFO] [1548533314.122602]: [/duckiebot2/cam_info_reader_node] ~image_type = compressed 
[INFO] [1548533314.138577]: [/duckiebot2/duckiebot_visualizer] Initialzing.
[INFO] [1548533314.154464]: [/duckiebot2/cam_info_reader_node] Using calibration file: /data/config/calibrations/camera_intrinsic/duckiebot2.yaml
[INFO] [1548533314.176435]: [/duckiebot2/duckiebot_visualizer] ~veh_name = duckiebot2 
[INFO] [1548533314.242484]: [/duckiebot2/duckiebot_visualizer] Initialzed.
[INFO] [1548533314.271685]: [/duckiebot2/cam_info_reader_node] CameraInfo: header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "/duckiebot2/camera_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [-0.2656611457928705, 0.0503147061229875, -0.00017810103535984586, -0.0038414674655902654, 0.0]
K: [322.76725793967654, 0.0, 311.09984601488344, 0.0, 320.397441492282, 240.50403477963104, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [221.51451110839844, 0.0, 308.8954407095116, 0.0, 0.0, 259.38458251953125, 240.57561591854028, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
[WARN] [1548533314.275258]: [/duckiebot2/cam_info_reader_node] ==============CompressedImage
[ERROR] [1548533315.201612140]: Couldn't open joystick /dev/input/js0. Will retry every second.
[INFO] [1548533315.748621]: [Stop Line Filter] ~stop_distance = 0.22 
[INFO] [1548533315.792402]: [Stop Line Filter] ~min_segs = 6 
[INFO] [1548533315.827264]: [Stop Line Filter] ~off_time = 2 
/home/software/catkin_ws/src/20-indefinite-navigation/stop_line_filter/src/stop_line_filter_node.py:40: UserWarning: '~max_y ' is not a legal ROS graph resource name. This may cause problems with other ROS tools
  rospy.set_param(param_name,value) #Write to parameter server for transparancy
[INFO] [1548533315.873252]: [Stop Line Filter] ~max_y  = 0.2 
[INFO] [1548533317.594405]: camera info topic is /duckiebot2/camera_node/camera_info
[INFO] [1548533317.595885]: waiting for camera info
[INFO] [1548533317.957318]: camera info received
[INFO] [1548533318.159979]: [Lane Filter] new filter config: ['lane_filter.LaneFilterHistogram', {'configuration': {'lanewidth': 0.23, 'range_min': 0.2, 'range_max': 0.6, 'd_max': 0.3, 'linewidth_white': 0.05, 'phi_min': -1.5, 'sigma_d_0': 0.1, 'curvature_res': 0, 'phi_max': 1.5, 'curvature_right': -0.054, 'mean_d_0': 0, 'sigma_d_mask': 1.0, 'mean_phi_0': 0, 'd_min': -0.15, 'linewidth_yellow': 0.025, 'curvature_left': 0.025, 'min_max': 0.1, 'range_est': 0.33, 'delta_d': 0.02, 'cov_v': 0.5, 'sigma_phi_mask': 2.0, 'sigma_phi_0': 0.1, 'delta_phi': 0.1}}]
[INFO] [1548533318.780874]: [/duckiebot2/joy_mapper_node] Initializing 
[INFO] [1548533318.821596]: [/duckiebot2/joy_mapper_node] ~speed_gain = 0.41 
[INFO] [1548533318.866648]: [/duckiebot2/joy_mapper_node] ~steer_gain = 8.3 
[INFO] [1548533318.940187]: [/duckiebot2/joy_mapper_node] ~bicycle_kinematics = 0.0 
[INFO] [1548533319.003910]: [/duckiebot2/joy_mapper_node] ~steer_angle_gain = 1.04 
[INFO] [1548533319.052257]: [/duckiebot2/joy_mapper_node] ~simulated_vehicle_length = 0.18 
[INFO] [1548533322.922690]: [/duckiebot2/velocity_to_pose_node] Initialized.
[INFO] [1548533323.192147]: [/duckiebot2/wheels_driver_node] Initializing 
Instance of simpleColorBalanceClass created.
[... 38 lines cut ...]
    self.impl.add_callback(callback, callback_args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 700, in add_callback
    raise ROSException("subscriber [%s] has been closed"%(self.resolved_name))
rospy.exceptions.ROSException: subscriber [/duckiebot2/lane_filter_node/lane_pose] has been closed
[Testcase: testkinematics_test_node] ... ok

[ROSTEST]-----------------------------------------------------------------------

[indefinite_navigation.rosunit-kinematics_test_node/test_drive_forward][FAILURE]
False is not true
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/home/software/catkin_ws/src/20-indefinite-navigation/indefinite_navigation/tests/test_kinematics.py", line 54, in test_drive_forward
    self.setup()
  File "/home/software/catkin_ws/src/20-indefinite-navigation/indefinite_navigation/tests/test_kinematics.py", line 42, in setup
    self.assert_(self.lane_message_received)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1`

liampaull, 2019-02-07:
@davesnowdon can you tell me where you saw the instruction to run this rostest?

davesnowdon, 2019-02-17:
@liampaull I was following the instructions on http://docs.duckietown.org/DT18/opmanual_duckiebot/out/wheel_calibration.html

liampaull, 2019-02-18:
ok thanks - I will look into this at some point - but it shouldn't block you from continuing on

liampaull, 2019-02-18:
@davesnowdon ok thanks - we should look into it but I don't think it blocks you from moving on. @tanij might also know more

closed 2019-04-12 by tanij