duckietown/Software #394: I can't reach the procedure to move Duckiebot

FUJI-AF, 2019-02-06:
I have error of docker.
what should I type command?
[image]

[image]

There are points I don’t know…
Especially I do not know UNIT B-7…
I attach the image of the error, I want some advice…
I am a beginner Linux…

FUJI-AF, 2019-02-12:
Please let me know if anything is missing.

liampaull, 2019-02-12:
if you do `export DOCKER_HOST=duckiebot.local` you no longer need to put the `-H duckiebot.local` (that's kind of the point of doing the first step). But I don't see why that would break things. This is a hard one to diagnose since I worry that it's networking issues. Can you be sure that you can ping the robot and retry?
(@breandan might have another suggestion)

FUJI-AF, 2019-02-19:
[image]

I can ping the robot.

breandan, 2019-02-19:
Did you try running `docker -H 192.168.11.21:2375 info`? What is the result? It's odd you can SSH via `ssh duckiebot` but ping using `ping duckiebot.local`.

FUJI-AF, 2019-02-19:
[image]

and I do not remember setting fujimoto@duckiebot.local's password. I can pass duckie@duckiebot.local's password.

FUJI-AF, 2019-02-19:
> Did you try running `docker -H 192.168.11.21:2375 info`? What is the result? It's odd you can SSH via `ssh duckiebot` but ping using `ping duckiebot.local`.

[image]

I try docker `-H 192.168.11.21:2375 info`. What should I read from here?

breandan, 2019-02-19:
There is some issue with the MDNS configuration on your laptop. If you can `ping duckiebot.local` but `docker -H duckiebot.local ...` fails, check `/etc/hosts`. Are there any entries here? Worst case, you can add this line manually: `192.168.11.21 duckiebot.local` or `192.168.11.21 duckiebot` and then you should be able to `ssh duckiebot[.local]`, `docker -H duckiebot[.local] ps`, and `ping duckiebot[.local]` successfully.

FUJI-AF, 2019-02-19:
[image]

Thank you, I did it

FUJI-AF, 2019-02-19:
[image]

I want to know why this is not found.

FUJI-AF, 2019-02-19:
> [image]
> 
> I want to know why this is not found.

https://github.com/duckietown/rpi-duckiebot-simple-python
this URL

liampaull, 2019-02-19:
This was an error in the documentation -  I have updated. The command should have been:

    $ docker -H <YOUR_DUCKIEBOT>.local  build -t my-simple-python-program .

closed 2019-02-19 by liampaull

FUJI-AF, 2019-02-19:
>  docker -H <YOUR_DUCKIEBOT>.local  build -t my-simple-python-program .

Thank you. Sorry for the low level question.

FUJI-AF, 2019-02-19:
[image]

dts commands become to failed..

liampaull, 2019-02-19:
Sorry to be a stickler but can you make a new issue in [this repo](https://github.com/duckietown/duckietown-shell)

FUJI-AF, 2019-02-19:
> Sorry to be a stickler but can you make a new issue in [this repo](https://github.com/duckietown/duckietown-shell)

OK