duckietown/Software #396: Apriltags detection

kc645, 2019-02-15:
I am trying to launch the apriltags2 detection packages.
However, when I try to view to image in rqt_image_view, the screen gives me a blank, grey image not displaying any images.
I checked out rqt_graph and 
[image]
this is what I see. What could be the solution?

kc645, 2019-02-15:
running the continuous detector launch file and looking at the rqt_graph makes this:

[image]

liampaull, 2019-02-16:
what is the command that you ran?

kc645, 2019-02-17:
i ran "roslaunch apriltags2_ros apriltag_detector_node.launch veh:='duckiebot1'" (for the first picture)
and "roslaunch apriltags2_ros continuous_detection.launch" (for the second picture)

liampaull, 2019-02-19:
I'm not sure that those launch files are meant to be run this way. Did you see that in the instructions somewhere? (@afdaniele might know more?)

kc645, 2019-02-19:
No I couldn't find any instructions so I used my previous knowledge of the apriltags. (I guess I was wrong).

liampaull, 2019-02-19:
In general, our convention is that the launch files that exist in the individual packages only launch the specific nodes in those packages. To get end-to-end functionality, have a look at the launch files in [here](https://github.com/duckietown/Software/tree/master18/catkin_ws/src/70-convenience-packages/duckietown_demos/launch) which all load the `master.launch` file, or for specific smaller functions in [here](https://github.com/duckietown/Software/tree/master18/catkin_ws/src/00-infrastructure/duckietown/launch). I haven't tried it but my best guess is that you should actually run:

    $ roslaunch duckietown apriltags2.launch veh:=duckiebot1

kc645, 2019-02-19:
will try that and let you know. Thank you!

kc645, 2019-02-20:
I ran 
roslaunch duckietown apriltags2.launch veh:=duckiebot1
and my rostopic list gave me
[image]
and
[image]
and rqt_graph gave me
[image]

and rqt_image_view still gave me
[image]

liampaull, 2019-02-20:
can you pull, rebuild and try again - for change see #399

kc645, 2019-02-20:
how do I do that? do I just "git pull" from inside the Docker image, or do i need to kill and re-initialize the base image?

liampaull, 2019-02-22:
Sorry for the delay - I have just pushed an updated image - so if you re-pull the image you should be able to retry. Thanks

kc645, 2019-02-27:
All resolved. Thank you for your help!

closed 2019-02-27 by liampaull