duckietown/Software #404: exiting dts shell error report

altineller, 2019-02-28:
Hello,

When I do dts start_gui_tools hostname, I get a shell, and everything works, but when I exit the shell by typing exit, I get the following error from stdout:

`root@borg:/home/software# exit
exit
Done!
ping: unknown host /bin/bash
Unable to locate /bin/bash /home/can/.dt-shell/commands/start_gui_tools/start_gui_tools.sh robi 10.10.10.103 host.local!
ping: unknown host /bin/bash

dts :  Traceback (most recent call last):
    :    File "/home/can/.local/lib/python2.7/site-packages/dt_shell/__init__.py", line 38, in cli_main
    :      cli_main_()
    :    File "/home/can/.local/lib/python2.7/site-packages/dt_shell/__init__.py", line 106, in cli_main_
    :      shell.onecmd(cmdline)
    :    File "/usr/lib/python2.7/cmd.py", line 221, in onecmd
    :      return func(arg)
    :    File "/home/can/.local/lib/python2.7/site-packages/dt_shell/cli.py", line 409, in <lambda>
    :      do_command_lam = lambda s, w: do_command(klass, s, w)
    :    File "/home/can/.local/lib/python2.7/site-packages/dt_shell/dt_command_abs.py", line 47, in do_command
    :      cls.command(shell, args)
    :    File "/home/can/.dt-shell/commands/start_gui_tools/command.py", line 35, in command
    :      start_gui_tools(script_cmd)
    :    File "/home/can/.dt-shell/commands/utils/docker_utils.py", line 278, in start_gui_tools
    :      duckiebot_ip = get_duckiebot_ip(duckiebot_name)
    :    File "/home/can/.dt-shell/commands/utils/networking_utils.py", line 19, in get_duckiebot_ip
    :      duckiebot_ip = get_ip_from_ping(duckiebot_name)
    :    File "/home/can/.dt-shell/commands/utils/networking_utils.py", line 11, in get_ip_from_ping
    :      raise Exception("Unable to locate %s!" % alias)
    :  Exception: Unable to locate /bin/bash /home/can/.dt-shell/commands/start_gui_tools/start_gui_tools.sh robi 10.10.10.103 host!

dts :  Please report that you are using:
    :  
    :  duckietown-shell: 3.0.31
    :  duckietown-shell-commands: 1.1.1
    :  python: "2.7.12 (default, Nov 12 2018, 14:36:49) \n[GCC 5.4.0 20160609]"`

tanij, 2019-04-12:
@altineller sorry for getting back to you so late. 

The shell has been updated. Are you still facing this problem?

liampaull, 2019-04-13:
confirmed and reproduced. It's an artifact of the way that docker exits. Low priority for now.