duckietown/Software #405: How to make changes in code inside duckiebot ?

arshdefoamed, 2019-03-15:
I have finished everything with respect to setting up the duckiebot, calibrating wheels and camera and running demos on the duckiebot. Now, I want to make changes to the logical code inside the duckiebot [Mostly different parameters that govern the movement of the bot]. However, I am not able to find any documentation that can help with that. Can someone explain how to do it?

The manual that I have been following so far is : 
https://docs.duckietown.org/DT19/opmanual_duckiebot/out/demos.html

liampaull, 2019-03-16:
Can you take a look at my answer [here](https://www.duckietown.org/forums/topic/running-ros-code-on-the-duckiebot#post-35094) and let me know if that works?

arshdefoamed, 2019-04-05:
I tried doing what you recommended. However, the terminology used was too technical for me. I did read about the -v  option but could not understand how that can be used. Could you please elaborate your approach?

tanij, 2019-04-12:
@liampaull do we have updated documentation on this?