duckietown/Software #406: docker container interaction

hosnerm, 2019-03-19:
Hi everyone 
i built two containers from the images _duckietown/rpi-duckiebot-base_ and _duckietown/rpi-duckiebot-joystick-demo_.

 when i try to `docker run` the joystick-demo on my duckiebot, i think that another launch file than the one that is in my base-container is executed. i think so because of 2 reasons:
1. if i access my base container and launch the joystick demo with `roslaunch duckietown joystick.launch veh:=hostname`, it works fine
 2. i deleted the forward_kinematics_node.py in my launchfile and the joystick-container crashes because of the forward_kinematics_node.py

to me it seems as if the joystick container doesn't read from the base image that i created. 
I adjusted the Dockerfile of the _duckietown/rpi-duckiebot-joystick-demo_ to be as follows (where devel-dbv2 is the tag of my own base container): 

[image]

If anyone can tell me how these containers interact (or why not) i would be very thankful.

liampaull, 2019-03-20:
it seems to me like this should work but it's easy to make mistakes in the image building process.

tanij, 2019-04-12:
@liampaull @hosnerm issue resolved?

liampaull, 2019-05-09:
the `joystick-demo` image is no longer used and everything should simply run off of base.

closed 2019-05-09 by liampaull