duckietown/Software #407: Error U-B-8-1

kazukikuroe, 2019-03-22:
I am trying to move a duckiebot.

Development environment is ubuntu18.04
Windows10

There.
$ docker -H hostname.local run -dit --privileged --name roscore --net host -v /data:/data --restart unless-stopped duckietown/rpi-ros-kinetic-roscore:master18

I entered the above command.Then I got this error.

Error response from daemon: OCI runtime create failed: container_linux.go:348: starting container process caused "exec: \"/ros_entrypoint.sh\": stat /ros_entrypoint.sh: no such file or directory": unknown.

I do not understand.

kazukikuroe, 2019-03-22:
`$ docker -H hostname.local run -dit --privileged --name roscore --net host -v /data:/data --restart unless-stopped duckietown/rpi-ros-kinetic-roscore:master18`

liampaull, 2019-03-24:
I have seen this error a few times very occasionally and never been able to figure it out. If you don't mind can you try and reflash the SD card and see if it goes away?

kazukikuroe, 2019-03-25:
I agree.
I will try to do so.
Thank you very much!!

tanij, 2019-04-12:
@kazukikuroe issue resolved?

liampaull, 2019-04-13:
@kazukikuroe can you retry - this should be fixed. Close if yes.

kazukikuroe, 2019-05-05:
Resetting this problem can solve it.

I was able to move the duckiebot by resetting it.

closed 2019-05-05 by liampaull