duckietown/Software #413: networkx 2.3 require python3 in requirement.txt

AmaurX, 2019-04-12:
@AndreaCensi  When trying to build the base container, it fails with the error:

```
Complete output from command python setup.py egg_info:
    NetworkX 2.3+ requires Python 3.5 or later (2.7 detected).
    
    For Python 2.7, please install version 2.2 using:
    
    $ pip install 'networkx==2.2'
```

I guess we should change the requirement.txt to match that?

liampaull, 2019-04-13:
@AmaurX what branch?

AmaurX, 2019-04-13:
@liampaull this happened both on master19 and master18.
On 11 of april, networkx released version 2.3 that no longer supports python2 (https://networkx.github.io/documentation/networkx-2.3/). As this is a ros container, we are still using python2 and therefore we should probably enforce using networkx==2.2 in the requirements.

liampaull, 2019-04-13:
ok cool - make a PR?

AmaurX, 2019-04-13:
My current pr on master18 includes it. Will make one for master19

liampaull, 2019-04-13:
yeah I just merged it

closed 2019-05-07 by liampaull