duckietown/Software #419: wheels_driver_node not in master19

bhairavmehta95, 2019-04-14:
Master18 branch, https://github.com/duckietown/Software/tree/master18/catkin_ws/src/00-infrastructure/duckietown/config/baseline/dagu_car

[image]

Master19 branch

[image]

Needed for lane control node?

bhairavmehta95, 2019-04-14:
Needed when launching `lane_controller_node` here: https://github.com/duckietown/Software/blob/master19/catkin_ws/src/10-lane-control/lane_control/launch/lane_controller_node.launch#L10

liampaull, 2019-04-14:
this is part of the refactoring that needs to be done of launch files in `master19`. All the drivers are now in the duckiebot-interface repo and will be launched separately.

bhairavmehta95, 2019-04-14:
Okay but short term, how do I launch this node then?

Necessary for [ROS Baseline](https://github.com/duckietown/challenge-aido_LF-baseline-duckietown/blob/aido2/lf_slim.launch#L56-L66)

liampaull, 2019-04-14:
right... I mean we never needed the wheels driver in the submission so this is actually why separating out the drivers makes sense. In other words, removing the launching of the wheels driver will make both things work - the demo and the competition submission. I can make the change.

liampaull, 2019-04-14:
https://github.com/duckietown/Software/issues/419#issuecomment-483051327
it is not needed when launching that launch file actually... very poorly coded but the wheels_driver isn't launched there

liampaull, 2019-04-14:
#420

closed 2019-05-07 by liampaull