duckietown/Software #428: RGB_led package not working

gibernas, 2019-05-13:
I tried to run a `rbg_led` script and got an error message of an unresolved import, which is mentioned as a comment
[here](https://github.com/duckietown/Software/blob/master19/catkin_ws/src/40-coordination/rgb_led/include/rgb_led/rgb_led.py) at line 28.

Moreover with the current structure, the `rgb_led`package should be moved to `01-interfaces`.

liampaull, 2019-05-13:
@gibernas how did you run this? In the interface container or in the base container?

liampaull, 2019-05-14:
@gibernas I confirm that it runs correctly from within the `duckiebot-interface` container but not from with a base demo container. Is this a problem?

liampaull, 2019-05-14:
ok I see the problem - the rgb_led package is imported by other things. I have added the adafruit drivers to the base container as a temporary fix. The following didn't work before but now does:

```
docker pull -H duckiebot.local duckietown/rpi-duckiebot-base:master19
dts duckiebot demo --demo_name base --duckiebot_name duckiebot
rosrun rgb_led fancy1
```
LEDs go crazy

closed 2019-05-14 by liampaull