duckietown/Software #436: Wheel Calibration Failure

doctorwho55, 2019-07-01:
I am unable to run the wheel calibration verification to ensure that my gain is calibrated properly.  I was directed to create an issue here by @liampaull .

### 1. Dependencies Issue
The first issue encountered is when opening the base container on the host computer as below.
`$ dts duckiebot demo --demo_name base --duckiebot_name duckiebot02`

This produces errors regarding unused args in a variety of files.  This is fixed by manually declaring the variables in the files in question.  This must be done with each new base container.  

Specifically, the `local` variable must be declared in the following files:

- `/home/software/catkin_ws/src/00-infrastructure/duckietown/launch/ground_projection.launch`
- `/home/software/catkin_ws/src/10-lane-control/lane_filter/launch/lane_filter_node.launch`
- `/home/software/catkin_ws/src/20-indefinite-navigation/stop_line_filter/launch/stop_line_filter_node.launch`
- `/home/software/catkin_ws/src/00-infrastructure/duckietown/launch/joystick.launch`

Also, the `param_file_name` variable must be declared in:
- `/home/software/catkin_ws/src/00-infrastructure/duckietown/launch/joystick.launch`

This issue has already been made apparent to Gianmarco Bernasconi via _Slack_, but is included here for completeness.

### 2. Calibration Failure
After resolving the first issue, the procedure for running the wheel calibration is as follows.

**IN FIRST TERMINAL WINDOW:**

Open base container:
`$ dts duckiebot demo --demo_name base --duckiebot_name duckiebot02
`

Then, within base container, launch kinematics calibration test:
```
root@duckiebot02:/home/software# roslaunch indefinite_navigation calibrate_kinematics.test veh:=duckiebot02
... logging to /root/.ros/log/c833a3de-9169-11e9-94ef-b827eb4ffefb/roslaunch-duckiebot02-239.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.202:40585/

SUMMARY
========

CLEAR PARAMETERS
 * /duckiebot02/forward_kinematics_node/
 * /duckiebot02/ground_projection/
 * /duckiebot02/inverse_kinematics_node/
 * /duckiebot02/joy_mapper_node/
 * /duckiebot02/lane_filter_node/
 * /duckiebot02/lane_pose_visualizer_node/
 * /duckiebot02/line_detector_node/
 * /duckiebot02/stop_line_filter_node/
 * /duckiebot02/velocity_to_pose_node/

PARAMETERS
 * /duckiebot02/duckiebot_visualizer/veh_name: duckiebot02
 * /duckiebot02/ground_projection/config: baseline
 * /duckiebot02/ground_projection/config_file_name: duckiebot02
 * /duckiebot02/joy_mapper_node/bicycle_kinematics: 0.0
 * /duckiebot02/joy_mapper_node/simulated_vehicle_length: 0.18
 * /duckiebot02/joy_mapper_node/speed_gain: 0.41
 * /duckiebot02/joy_mapper_node/steer_angle_gain: 1.04
 * /duckiebot02/joy_mapper_node/steer_gain: 8.3
 * /duckiebot02/lane_filter_node/filter: ['lane_filter.Lan...
 * /duckiebot02/lane_filter_node/use_propagation: True
 * /duckiebot02/line_detector_node/detector: ['line_detector.L...
 * /duckiebot02/line_detector_node/detector_intersection: ['line_detector.L...
 * /duckiebot02/line_detector_node/img_size: [120, 160]
 * /duckiebot02/line_detector_node/top_cutoff: 40
 * /duckiebot02/line_detector_node/verbose: False
 * /duckiebot02/stop_line_filter_node/max_y: 0.2
[... 104 lines cut ...]

**LOGS:**

Examine the logs (/root/.ros/test_results/rostest_kinematics_calibration/rosunit-kinematics_test_node.xml):
```
<?xml version="1.0" encoding="utf-8"?>
<testsuite errors="0" failures="1" name="unittest.suite.TestSuite" tests="1" time="10.392"><testcase classname="__main__.KinematicsTestNode" name="test_drive_forward" time="10.3918"><failure type="AssertionError">False is not true
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/home/software/catkin_ws/src/20-indefinite-navigation/indefinite_navigation/tests/test_kinematics.py", line 54, in test_drive_forward
    self.setup()
  File "/home/software/catkin_ws/src/20-indefinite-navigation/indefinite_navigation/tests/test_kinematics.py", line 42, in setup
    self.assert_(self.lane_message_received)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
</failure></testcase><system-out>&lt;![CDATA[
[INFO] [1561389014.720631]: wheel topic = /duckiebot02/joy_mapper_node/car_cmd
[INFO] [1561389014.721766]: lane topic = /duckiebot02/lane_filter_node/lane_pose
[INFO] [1561389014.722491]: stop_topic = /duckiebot02/stop_line_filter_node/stop_line_reading

]]&gt;</system-out><system-err>&lt;![CDATA[

]]&gt;</system-err></testsuite>

```

liampaull, 2019-07-03:
apologies for this - we are doing a bit of "on the fly" refactoring. I will look into it. Thanks for reporting

riacheruvu, 2019-07-14:
Hi! Has this issue been resolved yet? I am encountering the same issues when performing wheel calibration for my bot.